jakebullet70/OctoTouchController #19: License question

IzzySoft, 2022-10-10:
Could you please add information what license your app falls under? Without any license it's rather "all rights resevered" with "source available (to look at)" – while you hopefully meant it being free/libre open source :wink: Thanks in advance!

jakebullet70, 2022-10-10:
Really this is my way of giving back to the community. You can really do what you want with it except sell it. I will add license info in the next day or two.

jakebullet70, 2022-10-11:
Release on the same license as Octoprint itself.  - GNU Affero General Public License V3, i have updated this repo and also included the license file in the distro.

IzzySoft, 2022-10-11:
Wonderful! So let me give back, too: OctoTC will [become available here](https://apt.izzysoft.de/fdroid/index/apk/sadLogic.OctoTouchController) in my repo with the next sync around 6 pm UTC today. Be welcome to [pick a badge](https://gitlab.com/IzzyOnDroid/repo/#if-my-app-is-available-here-do-you-have-a-badge-i-can-use-to-link-to-it) to link there if you wish. I can also provide you the metadata (description, screenshots) in [Fastlane structure](https://gitlab.com/-/snippets/1895688) via PR, so you can control the presentation of your app in my repo yourself; just let me know if you want that.

Thanks a lot once more!

closed 2022-10-11 by IzzySoft