jakebullet70/OctoTouchController #31: OctoPrint Printerprofile inverted Axis are ignored

rmayergfx, 2023-03-05:
Inside Octoprint i have to invert the Y axis to match the movement.
OctoTC is connected to this instance with api, but ignored the settings to invert this axis.

So on the Display the movement is different to the setup.

Please add the option to invert the axis or read the flag from octoprint setup.

jakebullet70, 2023-03-06:
I will look at that in my morning.

jakebullet70, 2023-03-06:
Fixed, will be out in the next release this week.

rmayergfx, 2023-03-09:
Thx. it is working fine now on all axis.

jakebullet70, 2023-03-09:
Fixed

rmayergfx, 2023-03-09:
Sorry, needs to be reopened. I don´t know why, but after Printing and Homing e.g. from PC Webinterface the axis Y and Z are again inverted.

jakebullet70, 2023-03-09:
OK, thats strange, when OctoTouchController starts it reads the values for the inverted flag and saves them. It then uses those values. It never reads then a second time. I will try some more testing - I reopen this.

jakebullet70, 2023-03-09:
Are you saying though they are always working correctly from the Octoprint web interface?

rmayergfx, 2023-03-09:
Yes. from PC or any other device connected via Webbrowser it is always working fine. Find the issue after printjob was finished and i tried to move the bed forward to easy pull of the print and the printer was doing nothing, so i tried the "wrong" direction and the bed slides to me. so i don´t know why and how this is happening.  When it is wrong, also Z is wrong but Z is not inverted. Only x axis stays fine.

jakebullet70, 2023-03-10:
Found a place where there could be an issue: I am testing.

jakebullet70, 2023-03-11:
I have release V1.29 - We no longer read inverted information from Octoprint. This is now set up locally in the General Options screen in the app.

jakebullet70, 2023-03-15:
Is this now working?

jakebullet70, 2023-03-19:
fixed

closed 2023-03-19 by jakebullet70