jakebullet70/OctoTouchController #35: Red Cross on Printig page after start

rmayergfx, 2023-03-11:
i always get an red X on the right side after pressing the Printing Button.
Can you explain why? Pressing on this red X in the square changes to the temperature view.

How can i get off the red X together with octoprint or is it possible to set in the settings default to show the temperatur?

jakebullet70, 2023-03-11:
The Red X is no thumbnail.  Do you have thumbnails enabled in your slicer?

jakebullet70, 2023-03-11:
Good idea though to make it optional and go to the temp screen. I will add it as an option.

jakebullet70, 2023-03-11:
Oh, found a bug where the red X could get stuck (even if you have thumbnails on) depending on how the print was started. Fixed.

rmayergfx, 2023-03-12:
on which release?

jakebullet70, 2023-03-12:
I will be putting out a new release in the next few days. it will have the red x fix and the new bed leveling wizard.

jakebullet70, 2023-03-16:
V1.3.0 has been released.

closed 2023-03-16 by jakebullet70