jakebullet70/OctoTouchController #51: "Getting thumbnail..." keeps fading in and out

GoTVm, 2023-07-21:
Went into the printing menu, nothing was listed to print. Now "Getting thumbnail..." keeps fading in and out.

jakebullet70, 2023-07-21:
Can you see the thumbnails in Octoprint web UI?

jakebullet70, 2023-07-22:
Also what is your phone / tablet make and model?

GoTVm, 2023-07-22:
I don't see thumbnails in my Octoprint web UI. Did I forget to install something?

My tablet is an Asus K012.

jakebullet70, 2023-07-22:
Yep, you have to get it working in Octoprint 1st. See:
https://plugins.octoprint.org/plugins/prusaslicerthumbnails/

GoTVm, 2023-07-22:
Is there a way to disable thumbnails? I don't really need them.

jakebullet70, 2023-07-22:
No, I build it because I wanted thumbnails. :)  
I have a version 2 I am about ready to release, I will add a new option to this version to turn on / off thumbnail support.

Can you print at all or is it stuck trying to find a thumbnail?

jakebullet70, 2023-07-22:
Oh you found a bug, I am looking at my code and it should stop flashing "Getting thumbnail..." if it does not find them after 3 tries.

GoTVm, 2023-07-22:
> Can you print at all or is it stuck trying to find a thumbnail?

I have not tried printing yet, but it keeps fading in and out even after leaving and re-entering the menu. Everything works just fine.

> it should stop flashing "Getting thumbnail..." if it does not find them after 3 tries.
Yesterday it would keep flashing, now it doesn't even appear anymore. There's just the X on the thumbnail but the text is no longer showing.

jakebullet70, 2023-07-22:
X on the thumbnail when not found is normal. Honestly at this point I would just install the thumbnail plugin to Octoprint and all should work. 

https://plugins.octoprint.org/plugins/prusaslicerthumbnails/

You found an issues when not using thumbnails, but you are the only user to notice. I will have a fix in the next version but not sure on a release date yet.

jakebullet70, 2023-07-30:
#52 is this the problem you where having?

jakebullet70, 2023-07-30:
There is a new release out. V1.3.1 - closing

closed 2023-07-30 by jakebullet70