jakebullet70/OctoTouchController #52: Getting thumbnail keeps fading, then doesn't work at all

CristianEduardMihai, 2023-07-28:
As in https://github.com/jakebullet70/OctoTouchController/issues/51 , `Getting thumbnail` keeps fading when and after going into the printing menu. This seems to stop when starting a print, but then you see no thumbnail.
I use prusaslicer and have the Slicer Preview plugin. I can see previews when selecting files for printing
Here is a sample code (GitHub won't let me upload .gcode)
[calicat_0.4n_0.3mm_PLA_MK3S_40m.txt](https://github.com/jakebullet70/OctoTouchController/files/12194867/calicat_0.4n_0.3mm_PLA_MK3S_40m.txt)

Here is a video of the fading:
https://github.com/jakebullet70/OctoTouchController/assets/59600013/d22e452c-bec7-42f0-9a95-989d1f959fe2

And a photo of what the `Printing` menu looks like during printing
[image]

jakebullet70, 2023-07-29:
Thanks for the video. I will take another look.

jakebullet70, 2023-07-29:
If you touch the heat hotend / heat bed icons while printing, does it change to the thumbnail?

CristianEduardMihai, 2023-07-29:
https://github.com/jakebullet70/OctoTouchController/assets/59600013/07a8abb7-07c3-4dfb-b89f-87310c23fef1

Oh yea, that does it. My bad.
Guess that's half fixed. Thanks!

jakebullet70, 2023-07-29:
I will get a fix for the other issue in my morning.

CristianEduardMihai, 2023-07-29:
Thanks for such a quick action! Слава Украине!🇺🇦

jakebullet70, 2023-07-30:
OK, new version. If you go to the hamburger menu (upper left on main screen) there will be a check for new version option. Download the new version. Its 1.3.1 - You can see what version is installed in the About screen. 

Let me know if this fixes the problem. I never could reproduce it on my end, but I have not been 3d printing this year as access to parts and filament is difficult in a frontline city. There might also be some new dialogs / features. I had been working on an update earlier in the year but with the war i had to take a break and cannot remember what I was doing...   LOL. If all is working good, I will do an official release.

So... Let me know, if you have any features or ideas send them my way!

Slava Ukraini!

CristianEduardMihai, 2023-07-30:
Thanks so much!
It's fixed, no more fading pop-ups

As per features and ideas, I'll try to learn the language and do a PR on my own, ease your workflow a little bit :)

closed 2023-07-30 by CristianEduardMihai