jakebullet70/OctoTouchController #57: can't change filament while paused

lexelby, 2023-09-26:
First off, thanks a ton for this project!  It's exactly what I was looking for.

One thing I really would like to be able to do is to use my touch controller for filament changes mid-print.  My printer (ultimaker 2+) doesn't support M600, and using its built-in filament change menu results in print head crashes (it's really not designed to be used when you're printing via USB).

I got the filament change function all configured how I wanted it.  I have my slicer set to send @pause on filament changes so that octoprint pauses.  But it looks like OctoTouchController still considers my printer to be "operational" and so it tells me I can't change the filament, "printer is busy".

Any chance we can override that busy check when the printer is paused?

jakebullet70, 2023-09-27:
Well as I have the source code all ripped apart (this is a good thing) for Version 2 coming out soon this would be the *perfect* time to add your style of ultimaker filament change support. If your willing, I need some things from you.  (When I refer to OctoTC its just short for OctoTouchController)

First. What model phone / Tablet are you using?

So, if you are willing as I do not have a ultimaker.
1. A video of what exactly you see on the OctoTC screen.
2. What exactly are you changing in the GCode. Sounds like just a pause? I am assuming this is *your* work around for the lack of an M600?
3. You get to BETA test.  :)  I am sure there will be a few new versions I will compile and you will have to download and test.
4. The more information you can give me on your steps and you expected result from OctoTC the better.

That "operational" (all status's) comers from Octoprint. I think we might have to add a 'ultimaker filament' change option. 
I should finish my 'manual mesh' and 'Set Z offset' wizards code I am working on this week so could start on something for you next week sometime.

On a positive note. V2 includes lots of new features you can give feedback on.

Let me know what you think.

lexelby, 2023-09-27:
Sure, I can help!  I'll try to get that video for you tomorrow.

I'm not sure my situation is unique to ultimaker printers.  I think it would be the case for any printer that doesn't have m600 support.  I would think that anyone who _does_ have m600 support wouldn't bother using your manual filament change feature...?

It would also be really useful if I could do all of the movement commands during a pause.  I've set up my pause and resume scripts in octoprint to save and restore the position, so there's no risk of screwing up the print.  Basically, if you added an "allow movement commands while paused" option that would solve everything for me.

Looking forward to v2 :)

jakebullet70, 2023-09-27:
"allow movement commands while paused"???  Love it!  Yep, I can do that. 

Can I see your PAUSE / RESUME scripts? Once we get this working would like to add them as sample of NON M600 filament change to the wiki. I see your point as this can / should support other non M600 firmware.

Think about how you would like this non M600 process to flow? in the V2 PRINT screen there is a 'new' swipe right side menu that can shown.

My idea of my 'manual filament change feature' was more of using NON printing wise. Are you seeing / thinking something different?

 
[image]

[image]

jakebullet70, 2023-09-27:
In the 2nd picture above, wondering if we need a new menu option to take it through *your* method of changing filament?

jakebullet70, 2023-11-09:
This should be fixed in the V2 pre-release as you can now use the movement controls when the printer is paused.

lexelby, 2023-11-09:
Thanks, and sorry for not getting back to you with the video!

jakebullet70, 2023-11-11:
No worries. Hope it works for you. I will close this ticket, if you need more help let me know.

closed 2023-11-11 by jakebullet70