jakebullet70/OctoTouchController #58: Support https

svenstaro, 2023-10-01:
The code doesn't currently support https as it's hard-coded to http. Would be great to also support https.

jakebullet70, 2023-10-01:
As this project is aimed at re claiming old phones / tablets that you put next to your printer and never leave your home (hence why there is no support for a camera) I was not too worried about security. 

Is there something you are seeing I missed?

svenstaro, 2023-10-01:
Well, I just had my Octoprint on https-only and so I now had to configure it to allow http as well. Nowadays some webservers will default to https so for me it kinda makes sense to also support this. And don't worry: My phone for this is quite old!

Cool app by the way. A bit laggy for me (due to many prints in the list, looking to prune them somehow).

jakebullet70, 2023-10-01:
I am glad i saved an old phone! :) What version of Android does it run? Some Android 4.0 - 4.4 devices do not have the support for the latest HTTPS so not sure how to deal with that.

Working on version 2 now. I never really have allot of files in my list but you are the second person that mention that. (He got around it by creating sub folders and putting files in there, so he has them saved and ready if he needs them. I do not support sub folders yet) I am changing the way I deal with files in the new version so hopefully we should see a speed increase.

If you have any suggestion let me know as the code is being worked on actively daily at this point.

svenstaro, 2023-10-01:
I got around the laggy files list by installing a plugin that deletes old prints. This keeps the list neater anyway. Octoprint also isn't great at dealing with a massive list so this is fine for me. I'm on Android 9 and it should support current TLS standards for some time to come.

One problem I keep having which is unrelated to this particular issue but you asked: In landscape orientation, the app will sometimes get stuck on the splash screen with the logo. I'm not sure it's related to landscape orientation but that's the only time I noticed it. If I restart the app, it'll randomly work. This isn't much to go on, I know.

jakebullet70, 2023-10-01:
I will do some extra testing in landscape orientation to see if i can see any issue. Never seen it before.