jakebullet70/OctoTouchController #61: signature/signing certificate changed?

IzzySoft, 2023-11-09:
I just got a warning that the signature of the last release does not match the previously used, so **updates are impossible.** Checking the release notes, I found no hints on that. May I ask what happened?

Previous certificate:

```gradle
Signer #1 certificate DN: CN=Steven A De George SR, O=sadLogic, C=UA
Signer #1 certificate SHA-256 digest: fdd49e6487703c349ddaf5e4f0ddda874237645546a34045868de3e6029620cf
Signer #1 certificate SHA-1 digest: 15fca8330ce0f6872c3751c7231888e7f23d6c06
Signer #1 certificate MD5 digest: 593841bd9ac0002b4f2f7e70ca90614e
Signer #1 key algorithm: RSA
Signer #1 key size (bits): 2048
```

New certificate:

```gradle
Signer #1 certificate DN: CN=Steve De George, O=sadLogic, C=UA
Signer #1 certificate SHA-256 digest: 3e287452f7b5efd3e6e0fb90c4eee9f6943da3f9842a36b59fc91215c916a4cd
Signer #1 certificate SHA-1 digest: 6f9ea843dc76f4e4c4441a156033ff86a7fc0a7f
Signer #1 certificate MD5 digest: 08957e0cbc4a4cf5c901969d3508c01a
Signer #1 key algorithm: RSA
Signer #1 key size (bits): 2048
```

Slightly different DN (the "SR" was dropped from the CN), but definitely different certificates (which is logical, as you cannot change the DN without changing the entire cert).

jakebullet70, 2023-11-09:
[image]

Yes, there was a cert change as the way the saved data was stored was re-written so V2.0 was not compatible with V1.X saved data.

IzzySoft, 2023-11-10:
I see – thanks for the details! So just for confirmation, could you provide the same APK signed with the original key once (as a key change could also mean a "hostile take-over", see [here](https://f-droid.org/2023/09/03/reproducible-builds-signing-keys-and-binary-repos.html) for some background)? Once confirmed, I'd update metadata here accordingly to have the new signature accepted (if you wonder what I'm talking about: your app is [listed here](https://apt.izzysoft.de/packages/sadLogic.OctoTouchController/) in my repo).

jakebullet70, 2023-11-10:
Sure, how do you want the APK?
Also thanks for the 'reproducible-builds-signing-keys-and-binary-repos' info page. Some things in there I had not thought of. Going to change my workflow.

IzzySoft, 2023-11-10:
> Sure, how do you want the APK?

Whatever seems suitable to you. Maybe attaching it to the very same release as e.g. `OneTouchController_oldSig.apk` would be a good idea, so whoever else wants to verify can do that as well?

As for the article: Glad it helped you! I've learned a lot during that process as well – and really was surprised when I enabled the `AllowedSigningKeys` feature in my repo how many "violations" happened. Also for the reasons behind them – and how many Android developers seemed unaware of what signing is for and how it works (no offense meant to anyone). So it's a summing-up of what I found (and did not expect) which I'd need to append to regularly (like that it's not the best idea having a Github workflow for releases which creates a new debug key for signing each time :see_no_evil: – or that there are further ways of verification (like a TXT record in your own DNS, if that's protected with DNSSec – an idea I got from one developer)). Learning new things each time!

Btw, while on it and speaking about workflows: would you consider establishing Fastlane structures here in your repo, so you can maintain app description and graphics to be used e.g. in my repo? I could open a PR submitting what is currently established here, and you could use my [Fastlane Cheat Sheet](https://gitlab.com/-/snippets/1895688) for guidance to build on that, e.g. adding per-release changelogs. Plus, if you like, you could [pick a badge](https://gitlab.com/IzzyOnDroid/repo/#if-my-app-is-available-here-do-you-have-a-badge-i-can-use-to-link-to-it) to link to your app in my repo if you wish, so folks interested find it there (together with other useful apps) with an easy way to install and receiving update notifications on new releases for all of them.

jakebullet70, 2023-11-11:
I added a new APK of V1.X to the release. The signature will match the old one.
As for Fastlane, unfortunately my time is limited being in the middle of a war. This V2 release was ready almost 3 months ago before I got time to get it uploaded.

IzzySoft, 2023-11-11:
> I added a new APK of V1.X to the release. The signature will match the old one.

The signature might match, but the APKs don't (1000+ files differ – for obvious reasons). I'll need the APK with `versionCode: 51` (i.e. the one originally attached to that release) but signed with the old key. Anyone can pick an older release and sign it with their key, but only the "keyholder" can sign a newer APK with the original key. The latter is a proof – the former proves nothing, sorry.

As for Fastlane: Once set up it takes little to no maintenance. You'd only need to update the description if it no longer matches (e.g. describes a feature you had to drop), or replace screenshots after a redesign – both things rarely happen. So the only thing for "each release" would be the per-release changelogs ("what changed with this release") – and while those are nice-to-have, they are not mandatory but simply "recommended" IMHO. Only about every 2nd app with Fastlane in my repo also has those changelogs.

But up to you: My offer (to send you a PR with what I have) stands, I won't "force" it upon you but you can call me upon it anytime :smile:

IzzySoft, 2023-11-14:
@jakebullet70 any chance for a ` OneTouchController_2.X_oldSig.apk`, so I can check and re-enable updates (with the new signature then)?

jakebullet70, 2023-11-14:
I do plan on getting it to you. Electricity / internet has been erratic this last week and we are still under sporadic heavy shelling..
Living in a frontline city in the middle of a war is an adventure.  
https://www.bbc.com/news/world-europe-67308647

IzzySoft, 2023-11-14:
Ouch, yeah… Sad to hear that :cry: Wishing you peace and freedom first – and soon at that!

jakebullet70, 2023-11-16:
OK, lets try this one. Its the 2.X code compiled with the old key.

IzzySoft, 2023-11-18:
Hm, it looks like GMS (a non-free dependency) slipped in with the last release. Was that intended, @jakebullet70? My "nightly scanner" complained.

jakebullet70, 2023-11-18:
Wow, I had to look up what GMS was. 
No it was not intended. I will try and take a look today. The only thing I can think of was I upgraded my toolchain from B4X 11 to B4X 12. 

Yesterday was 18+ hours of no electricity. Hopefully today will be better.  :)

jakebullet70, 2023-11-18:
" I'll disable the old key here then as I guess you won't use it anymore with newer releases (can you confirm?). "  
Yes, the old key is retired. Will not be used it anymore.

IzzySoft, 2023-11-18:
> No it was not intended. I will try and take a look today.

Thanks! Thought so. If you'd use Gradle, things like `gradle :app:dependencies` would come in handy to find the culprit having dragged that in. Not sure if there's something comparable to what you se for building (I'm no dev, so I cannot tell).

> Yesterday was 18+ hours of no electricity. Hopefully today will be better.

Oh my. Yeah, it can only get better. Not much room left for things getting worse :disappointed:

> Yes, the old key is retired. Will not be used it anymore.

Thanks! Applied the corresponding "mark" to the metadata here so I'll get a reminder in time.

jakebullet70, 2023-11-20:
OK got some time to take a look. Yes I added it in. (Do not remember adding it in but it was me!  LOL )
I switched from a REST API to sockets in version 2 as a sockets connections were needed to support some advance features in Octoprint.

It is only needed in Android V4.X  as Android V4.X does not support modern SSL. We do a patch to a security provider for modern compatibility. This is all the code is used for. We do not connect to the play store or anything like that.

Anyway, The easy fix is to drop support for Android 4 and remove this but I do not want to do this as while Android 4 maybe uncommon in the West it is still very common here in Eastern Europe and in fact I have 2 devices running Android 4.

-------------------------------------------------------------------------------------------------------
So... I am thinking I could produce 2 versions.  One that is Android V5 and above you can grab and one that is Android 4 and above just on GitHub.

I am open to any ideas?

[image]

IzzySoft, 2023-11-20:
Ah, now I get it – SSL. Yeah, there's that one Google lib for it. I think I've heard about a FOSS alternative to it a while ago. If I only could remember its name. There were thoughts back then to include that with F-Droid, so other apps could simply build on it. If we can find that, you probably could use it with OTC, making separate versions unnecessary. What was the name of that component? I'm just having a blackout currently. With the name, I could try to find it (no promise of success, though). Hm, but the one I remember was not "SSL Websocket Client". It was something replacing the underlying SSL stuff.

If we cannot find a FLOSS replacement, two separate versions sound reasonable. I'd suggest different package names then, renaming the 4.x variant by e.g. using a `.legacy` suffix (`sadLogic.OctoTouchController.legacy`). 4.x users will rather be the minority (:wave: I still have a 4.4 device up and running as night stand, a Milestone 2 from 2010 – and I could well imagine others using their "old devices" for purposes like this). Then have a note in the app description for 4.x users to switch to that. As I'm a supporter of sustainability ("too good to throw away", "keep it running as long as you can" – my private main driver still is a 2015 Fairphone 2, running Android 11), I'll happily host both variants in my repo then. Or if you prefer to keep the package name (for easy updates), the variant you're proposing (keeping the 4.x variant "Github only") would of course be fine with me, too. Your choice :wink:

And thanks!!!

jakebullet70, 2023-11-22:
I thinking I will go with 2 releases (Legacy and normal) - Wondering if I need to generate a new key??? (my brain is frozzen! LOL) Feature wise there is no difference. In my research there is a allot of conflicting information on the TLS 1.0 - TLS 1.2 support on Android 4.X devices and it looks that it is manufacture dependent with Android 4.0 not being included in some of the proposed fixes.

One of the reasons this project was born was to keep perfectly good devices from the trash. I am not going to let a rich corporation (Google) tell me when to buy a new device.

Right now 1/2 the city has no heat. As soon as I can find some warmth and my brain unfreezes I will modify the build scripts and get that done.

IzzySoft, 2023-11-23:
> Wondering if I need to generate a new key???

You can use the existing key for both packages, no issues. If you want to refresh your background on this topic (and possible implications), I can recommend you an article I've published recently: [How to keep your key safe and what measures to take for the event of loss?](https://f-droid.org/2023/09/03/reproducible-builds-signing-keys-and-binary-repos.html#lessons-learned-2-how-to-keep-your-key-safe-and-what-measures-to-take-for-the-event-of-loss) (link goes to that specific section, you're of course welcome to read the entire thing).

> One of the reasons this project was born was to keep perfectly good devices from the trash.

You're running to a wide-opened door there with me. I often get asked why I serve the armeabi APK from apps having an arm64 APK available – same reason: most 64bit devices (with only 1 current and 1 upcoming exception known to me) still support 32bit, and one manufacturer even found a way to do that with a chip ARM declared to be "64bit only" ([see here](https://phandroid.com/2023/10/31/xiaomi-has-managed-to-get-32-bit-apps-to-run-on-the-snapdragon-8-gen-3/)). True, one possibly shouldn't use a device with outdated firmware+AndroidVersion as "daily driver" (especially from a security point-of-view), but there are plenty of use-cases they perfectly fit. Like my 2010 Motorola Milestone 2 (running Kitkat thanks to CyanogenMod) serving as a night-stand, only having needed a battery-replacement once yet, in 2022, i.e. after running 12 years with the original one). So why throw them away? This is also why I offered hosting both variants of your app with my repo, if they use dedicated packageNames.

> Right now 1/2 the city has no heat.

Ouch. That is hard :cry:

> As soon as I can find some warmth and my brain unfreezes I will modify the build scripts and get that done.

Thanks! Looking forward to that. Warm wishes (doubly so in this context) for now then!

jakebullet70, 2023-11-26:
On a good note heaters are coming back on in our part of the city. While I made the needed changes a while back I could not test as it was too cold to run the 3d printers. Hopefully we can get this issue tested in the next week or so an I can be warm again!  :)

jakebullet70, 2023-11-29:
I have release V2.0.1
Included is a FOSS version that has had the package name changed. I did it this way as releacing a new legacy version would of broken anyone who then updated if they where running a Android 4 device.  

I am now going back to getting warm. LOL

closed 2023-11-29 by jakebullet70

IzzySoft, 2023-11-29:
Thanks! Established the FOSS version in parallel and added a hint to the non-FOSS one (becomes visible with the next sync around 7 pm UTC). Both will be updated in parallel in the future. If you want me to add a hint to the FOSS version as well (for 4.x users), please let me know. Also if you'd like me to open a PR with the metadata in fastlane format, so you can maintain it yourself (see my [Fastlane Cheat Sheet](https://gitlab.com/-/snippets/1895688) for guidance).

Btw, funny that [VT](https://www.virustotal.com/gui/file/a66636758bddab744d0b6bde265163e4c078e6685f436e01f38b8d86f1d173bb/detection/f-a66636758bddab744d0b6bde265163e4c078e6685f436e01f38b8d86f1d173bb-1701249928) has 2 engines flagging the FOSS version, but none for the non-FOSS built from the same commit. Especially the engine called "Google" with its detailed hint "Detected" (but not naming what it detected). The other one is Ikarus thinking it found a trojan spy :roll_eyes: False positives, but I though you might like to know.

Enjoy the returning warmth! :sun_with_face: