jakebullet70/OctoTouchController #63: Question on permissions

IzzySoft, 2025-04-13:
Today's release yielded some warnings from our scanners:

```
! repo/sadLogic.OctoTouchController_55.apk declares sensitive permission(s):
  android.permission.REQUEST_INSTALL_PACKAGES android.permission.READ_EXTERNAL_STORAGE*
! repo/sadLogic.OctoTouchController.foss_55.apk declares sensitive permission(s):
  android.permission.REQUEST_INSTALL_PACKAGES android.permission.READ_EXTERNAL_STORAGE*
```

May I ask what those permissions are used/needed for – especially `REQUEST_INSTALL_PACKAGES`?

jakebullet70, 2025-04-14:
No worries. :)
Surprised your scanners did not pick this up before as the update feature has been there for awhile.

As the program was never designed to be in a 'App Store' it checks for its own updates and will prompt - request for an upgrade. 
This was also needed as the App supports Android 4.x deviece that were not patched and does not support the latest SSL.

https://github.com/jakebullet70/OctoTouchController/blob/V2.X/OctoTouchController.b4a
Around line 644 is the code that requests if you want to install an update.

https://github.com/jakebullet70/OctoTouchController/blob/V2.X/src/dialogs/dlgAppUpdate.bas
Around line 55 does the version checking.

In a nutshell. 
About once a month the App will check for an update and ask if you want to install it.

IzzySoft, 2025-04-14:
> Surprised your scanners did not pick this up before as the update feature has been there for awhile.

It also has been a while since your last release 😉 Those scanner features have been implemented around 2/2024, which was *after* the previous release here, see [Ramping up security: additional APK checks are in place with the IzzyOnDroid repo](https://android.izzysoft.de/articles/named/iod-scan-apkchecks).

> it checks for its own updates and will prompt - request for an upgrade.

That's what I was afraid would be the case – as it might violate the [IzzyOnDroid App Inclusion Policy](https://gitlab.com/IzzyOnDroid/repo/-/wikis/Inclusion-Policy), quote:

> [the app] must not download additional executable binary files (e.g. addons, auto-updates, etc.) without explicit user consent. Consent means it needs to be opt-in (it must not be harder to decline than to accept or presented in a way users are likely to press accept without reading) and structured in a way that clearly explains to users that they’re choosing to bypass the checks performed in this repo if they activate it.

(same criteria on F-Droid.org as well btw). So:

* update check needs to be opt-in, or it would raise the `NonFreeNet` anti-feature (assuming it reaches out to e.g. Github)
* the update itself must not only be opt-in, but also clarify the implications and where the APK is coming from: loading the APK from outside our repo means our additional screening/checks will be bypassed. If this point is not met, we cannot keep the app in our repo 😢

So are these two points met? I cannot see any setting being queried at the update check (something like "if updatecheck_enabled"), just the "time frame". And for the install itself, the code you pointed to just makes sure an installation can take place (activate "unknown sources" etc). I didn't find where it asks whether to install "this" update, or what implications it has (but "didn't find" might be due to lack on my side: I'm not familiar with Gcode at all, though it reminds me of Basic – which was fat too long ago on my end (30+ years)…)

```bas
Public Sub Start_ApkInstall(args() As String)
	Wait For (CheckInstallationRequirements) Complete (Result As Boolean)
	If Result Then
		SendInstallIntent(args(0))
	End If
End Sub
```

suggests the install will just happen automatically if it's possible (i.e. if the permission was granted in the past). AFAIK Android will then pop up a (generic) prompt (so one could skip an update there), but that would not include outlining the implications – or even just informing where the update is coming from.

So if my assumptions are correct here, may I kindly ask you to make the checks opt-in – and have the updates themselves outlining the implications plus requiring a confirmation when they happen? Thanks in advance!

jakebullet70, 2025-04-14:
Yes, it has been awhile. :) Living in Kherson, Ukraine is challenging to say the least. Well, we are not dead but our home does have a few extra holes in it.  

OK to clarify what the app does now  
An update is **NEVER** automatically installed. You are just notified there is one when the App starts.
**You have to then go to the setup menu and do the update manually**  
So really the user, unless they go to the setup menu and press 'Update' will NEVER get an update.  
The updates do come from github.

I can add a 'Do you want to check for updates' option that can be set at install?

Does F-Droid.org do Updates? 

-------------------------------------------------------------------------------
Language wise Github is wrong at what it thinks it is.
Program is written in B4X: https://www.b4x.com/
Looks like BASIC but it is modern with async events, classes and threads, ends up being transpiled into JAVA and compiled.

IzzySoft, 2025-04-14:
> Yes, it has been awhile. :) Living in Kherson, Ukraine is challenging to say the least. Well, we are not dead but our home does have a few extra holes in it.  

Oh my… Not meant sarcastic – but good to see you can take it with a pinch of humor ("extra holes"). Still sad that situation drags on – and that it started at all. This is not about people, just about power – which contradicts my principles. Why do people think land belongs to them? Don't we all belong to the earth? All so upside down…

Thanks for all your details there, Steven! The "do you want to check for updates" option would be a very good start. And it would be perfect if it includes some explanation, like: "Do you want to check for updates? This is done at the app's Github repo. If updates are available, you can then download and install the APKs directly from the settings menu. If you originally installed the app from a repository that performs extra checks/screenings (like IzzyOnDroid), please be aware that those screenings would then be bypassed." That would then fully meet the inclusion criteria again.

> Does F-Droid.org do Updates?

I was not aware your app would be on F-Droid.org as well (just checked: does not look like). But yes, both F-Droid.org as well as IzzyOnDroid perform automated updates. Here is what it says at IzzyOnDroid:

```yaml
AutoUpdateMode: Version V%v
UpdateCheckMode: Tags
```

Which basically means: "Look at the releases at Github, watch out for tags starting with `V`, and compare what follows with the newest release we have here. If Github has a newer release, get it (scan it etc, and finally provide it in the IoD repo)."

> Looks like BASIC but it is modern with async events, classes and threads, ends up being transpiled into JAVA and compiled.

Ah, so my guess with BASIC was basically right, nice to know, thanks! And yes, saw those async parts 😉

Btw, some suggestion for the shields on your Readme:

```html
<p align="center">
  <a href="https://github.com/jakebullet70/OctoTouchController/releases"><img src="https://img.shields.io/github/v/release/jakebullet70/OctoTouchController?logo=github&logoColor=white" alt="GitHub release"/></a>
  <a href="https://apt.izzysoft.de/packages/sadLogic.OctoTouchController.foss"><img src="https://img.shields.io/endpoint?url=https://apt.izzysoft.de/fdroid/api/v1/shield/sadLogic.OctoTouchController"/></a>
  <a href="https://github.com/jakebullet70/OctoTouchController/issues?q=is%3Aissue%20state%3Aclosed%20sort%3Aupdated-desc"><img src="https://img.shields.io/github/issues-closed-raw/jakebullet70/OctoTouchController"/></a>
  <a href="https://github.com/jakebullet70/OctoTouchController/issues?q=is%3Aissue+is%3Aopen+sort%3Aupdated-desc"><img src="https://img.shields.io/github/issues/jakebullet70/OctoTouchController"/></a>
</p>
```

That would make a click/tap on a badge link to where the indicated "thing" can be found 😉

jakebullet70, 2025-04-16:
OK, Power is back on. This is the city we live in. Not cunductive to doing development. 
https://www.reuters.com/world/europe/ukraines-most-dangerous-city-craves-respite-killer-drones-peace-talks-run-2025-03-24/

Anyway...
That sounds good. 
I am testing (when the power works) another update now and will add what you sugested for the next release. Should be in a week or so. I will keep this ticket open so we can chat if I have any more questions.

Thanks for the info on the shields, I have already added that.

IzzySoft, 2025-05-07:
Too sad B4x depends on Windows, so we cannot offer [Reproducible Builds](https://reproducible-builds.org/) for your apps  (see e.g. [Reproducible Builds, special client support and more at IzzyOnDroid](https://android.izzysoft.de/articles/named/iod-rbs-mirrors-clients) for some background)…

OK, I've performed a quick scan on HomeCentral. Same issue with `android.permission.REQUEST_INSTALL_PACKAGES` there. So I need to raise the same question again: "informed consent". Is the update check opt-in, and does it make clear the implications of bypassing the additional screening performed at IzzyOnDroid?

Really nice idea btw, especially the "giving old hardware a new life" 🤗 As it's especially for the kitchen, I was immediately looking for "shopping list" and "reminders" (leaving notes for other "home mates"). So if you're looking for ideas… 😉

IzzySoft, 2025-05-10:
> Let me know if that looks OK. This code runs on 1st install and is Opt-In. :)

Perfect!

> Funny you should talk about:
> "I was immediately looking for "shopping list" and "reminders" (leaving notes for other "home mates")"

It hangs in the kitchen, so that was the first thing coming to mind 😜

> The project (HomeCentral) is a refactor of a larger project I did in 2016 that does have a shopping list and house mate reminders. So... LOL. I already have that code, I just did not put it in. ;)

Cool!

> And as far as Reproducible Builds, there is a way to do builds in Linux. I would have to look it up but seem to remember someone on the B4X forum doing it.

I happily give it a try then – as long as it's possible from the command line, and does **NOT** involve Wine (all instructions I found so far depended on Wine, and that won't work for our RB process).

jakebullet70, 2025-05-12:
Moved the Opt In code over to this project so now we are good. :) On 1st run you get the opt in popup.
So Released 2.1.0

I am going to close this.

Also, yeah, all the B4X methods of doing Linux builds do require Wine.  :(

closed 2025-05-12 by jakebullet70

IzzySoft, 2025-05-12:
> Moved the Opt In code over to this project so now we are good. :) On 1st run you get the opt in popup.
> So Released 2.1.0

Wonderful! So I've put this to the "green list", which is shown with the repo browser in the package details on the permissions:

```yaml
android.permission.REQUEST_INSTALL_PACKAGES: for the integrated self-updater, which is strictly opt-in
android.permission.WRITE_EXTERNAL_STORAGE: to save the downloaded APK for install
android.permission.READ_EXTERNAL_STORAGE: to read the downloaded APK for install
```

That correct so (for the storage permissions)? No complaints anymore now from our scanners:

```
$ iod repo get sadLogic.OctoTouchController.foss
sadLogic.OctoTouchController.foss: looking for 'https://api.github.com/repos/jakebullet70/OctoTouchController/releases'
sadLogic.OctoTouchController.foss: checking tag 'V2.1.0'
sadLogic.OctoTouchController.foss: lastRelNo set to '2.1.0', checking for files
sadLogic.OctoTouchController.foss: Upstream file date (2025-05-12 07:18) is newer than ours (2025-04-13 19:41).
sadLogic.OctoTouchController.foss: returning ['2.1.0','https://github.com/jakebullet70/OctoTouchController/releases/download/V2.1.0/OctoTouchController_foss.apk',1747027110]
sadLogic.OctoTouchController.foss: 2.0.4/2.1.0, https://github.com/jakebullet70/OctoTouchController/releases: https://github.com/jakebullet70/OctoTouchController/releases/download/V2.1.0/OctoTouchController_foss.apk
- Grabbing update for sadLogic.OctoTouchController.foss: OK
- Checking 'repo/sadLogic.OctoTouchController.foss_56.apk' for libraries and malware …
- Checking the app's AndroidManifest.xml …
sadLogic.OctoTouchController.foss: check if repo contains FUNDING.yml
sadLogic.OctoTouchController.foss: looking for 'https://api.github.com/repos/jakebullet70/OctoTouchController/contents/.github'
sadLogic.OctoTouchController.foss: Github reports "Not Found" for https://api.github.com/repos/jakebullet70/OctoTouchController/contents/.github
sadLogic.OctoTouchController.foss: looking for 'https://api.github.com/repos/jakebullet70/OctoTouchController/contents/'
sadLogic.OctoTouchController.foss: looking for 'https://api.github.com/repos/jakebullet70/.github/contents/'
sadLogic.OctoTouchController.foss: Github reports "Not Found" for https://api.github.com/repos/jakebullet70/.github/contents/
sadLogic.OctoTouchController.foss: no FUNDING.yml detected.
sadLogic.OctoTouchController.foss: no Fastlane configured, skipping Fastlane check.
```

If you wonder about [FUNDING.yml](https://docs.github.com/en/repositories/managing-your-repositorys-settings-and-features/customizing-your-repository/displaying-a-sponsor-button-in-your-repository): you can add that if you want donation links to be generated. And as for Fastlane: that is so you can define how your app is presented  (description & graphics). If you're interested, I can send you a PR with a starter package, containing what's set up at our end (no big deal, takes me less than 5 minutes) – and you're welcome to use the [IzzyOnDroid Fastlane Documentation](https://gitlab.com/IzzyOnDroid/repo/-/wikis/Fastlane) as guidance/reference to build upon that.

> Also, yeah, all the B4X methods of doing Linux builds do require Wine. :(

And there I was hoping I might have missed something. Sad; then we have to forego reproducible builds 😢

jakebullet70, 2025-05-13:
Looks good. 
I did add the fastlane stuff but I see in the logs it did not get picked up. What did I do wrong?

[image]

jakebullet70, 2025-05-14:
OK, renamed the folder to phoneScreenshots and full_description.txt should be markdown friendly now. :)
Lets give it a try.

Also did the same to
https://github.com/jakebullet70/HomeCentral

IzzySoft, 2025-05-14:
> renamed the folder to phoneScreenshots

Thanks! Will activate them ASAP.

> and full_description.txt should be markdown friendly now.

Not yet, the newline before the bullet-point list is still missing:

```markdown
Features:  
 - Native Android app, not a re-packaged web app  
 - Complete support for OctoKlipper  
```

should be

```markdown
Features:  

- Native Android app, not a re-packaged web app  
- Complete support for OctoKlipper  
```

Also, better no leading spaces before each bullet point. I guess this single one would be ignored, but multiple would definitely cause issues.

A further question is how we use the same Fastlane for both variants while still making clear which variant one is looking at. Maybe give each one a dedicated name to refer to, e.g. name the 4.x one "OctoTC Legacy" or "OctoTC Classic"?

> Also did the same to https://github.com/jakebullet70/HomeCentral

Same issue with fulldesc there. Oh, I see we didn't include that yet here. As we have no open issue for that on our end, let's do it from here once fastlane is "ready" (<= note-to-self).